CoretechR/ZeroBug #5: Raspberry PI software does not work

kenking2536, 2021-12-25:
Raspberry PI software does not work

I'm not familiar with Raspberry PI. Please guide me how to install it,

Copy what files to where??

What documents and contents are modified??

I'm using Raspberry PI zero 2 W

Not Raspberry PI zero w

After many days of hard work! Learned a lot of knowledge! Finally, this problem occurs! Please have a look!
My zero 2 W installation process is as follows:

sudo apt-get install subversion
sudo apt-get install imagemagick
sudo apt-get install cmake
sudo apt-get install git
sudo apt-get install libv4l-dev
sudo apt-get install libopencv-dev
sudo apt-get install gcc g++
sudo apt-get install libjpeg9-dev
sudo apt-get install libjpeg62-turbo-dev
sudo apt-get install xrdp
sudo apt-get install samba
sudo apt-get install nodejs
sudo apt-get install npm
npm install socket.io
pip3 install aiohttp
pip3 install aiodns
pip3 install cchardet
pip3 install asyncio
pip3 install pygame
sudo git clone https://github.com/jacksonliam/mjpg-streamer
cd mjpg-streamer/mjpg-streamer-experimental
sudo make all
sudo make install
/home/pi/mjpg-streamer/mjpg-streamer-experimental/./mjpg_streamer -i "input_raspicam.so -x 1280 -y 720 -fps 12" -o "output_http.so -w ./www"
sudo cp mjpg_streamer@.service /etc/systemd/system
cd /etc/systemd/system
sudo mv mjpg_streamer@.service mjpg_streamer.service
sudo nano /etc/systemd/system/mjpg_streamer.service
sudo systemctl start mjpg_streamer.service
sudo systemctl enable mjpg_streamer.service
vcgencmd get_camera

Finally, the following error is displayed:
Traceback (most recent call last):
File "/home/pi/ZeroBug/app.py", line 2, in
import socketio
ModuleNotFoundError: No module named 'socketio'

pageauc, 2021-12-25:
> sudo cp ***@***.*** /etc/systemd/system

You might try

sudo apt install socket.io

rather than rpm install

kenking2536, 2021-12-25:
socket. IO problem solved!

thank!

Universal sudo apt-get install