CoretechR/ZeroBug #9: Servo Motor Replacement: EMAX ES08A II vs. Tower Pro MG90S

BiyolojiMeb, 2024-12-25:
Hello,
I can't find the EMAX ES08A II servo motor. Would it be okay if I use the Tower Pro MG90S instead?
Thanks!

CoretechR, 2024-12-25:
Hi, you should be able to use any micro servo, but you have to check the exact dimensions. The 3D-printed parts and especially the screw holes will probably not fit without some modifications. 
Generally, I would recommend using the ES08A or a higher quality servo. The MG90S are not as fast and precise as the ES08A.

tianyi725, 2024-12-28:
> Hi, you should be able to use any micro servo, but you have to check the exact dimensions. The 3D-printed parts and especially the screw holes will probably not fit without some modifications. [...]

thankyou